IEC-61400/power-curve-schema #5: Transfer ownership to common organisation

thclark, 2023-06-15:
## Feature request

### About me

I'm the CEO of Octue, and I wrote the first draft of this schema.

### Use Case

We're currently hosting this schema on behalf of WindPioneers and the IEC group who are collaborating on it. The IEA Wind Task 43 digitalisation group is overlapping and actively interested in helping.

Should we be transferring this repository to an IEC owned organisation? An alternative is to transfer it to [IEA Task 43's GitHub page](https://github.com/IEA-Task-43) alongside the [WRA data standard](https://github.com/IEA-Task-43/digital_wra_data_standard).

### Current state

Please describe what you're doing presently to work around this or achieve what you're doing.

Dash16PM, 2023-09-05:
I believe the decision was to leave the schema here for the near future, since no obvious home has been identified.

thclark, 2024-04-23:
A quick update: Talks ongoing with IEC to create a GitHub account; we can transfer it there if need be.