IEC-61400/power-curve-schema #13: Include bility to record measured power curves

thclark, 2023-07-06:
## Feature request

### Use Case

### Current state

thclark, 2023-07-06:
Notes on barinstorming

Alan derrick - says we should. IEC Standard 61400-12-1 defines the parameters required so we could derive a schema to match.
Stephen - we could use a 'type parameter to.
Alan - Need to include uncertainty in measured power curves.
Emil - not always an apples-to-apples comparison. We would need to state the filtering applied. Data looks like a label or comments.

kersting, 2023-08-03:
@thclark I am supportive of this idea, but we have to account for the fact that standards change with time. Therefore, a definition from 61400-12-1 may have changed on 61400-12-2 and it may change once 61400-12-3 is released. This is something that is biting us in IEA task 43 for the data model. We included an option in the sensor description for adding the anemometer class, but the anemometer classes have change when the standard was updated and the same anemometer can have different classes depending if one is using 61400-12-1 or 61400-12-2.

This problem may be already a concern in the current schema that you shared. In `design_parameters` we have `iec_class`. For me that should be a dictionary including `iec_standard_version` so one can include the version of the standard. Should I create another issue separated for this specific problem?

I'd tackle this problem first. Even if 12-1 and 12-2 are identical, it does not guarantee -12-3 will be. I can support this effort too.

Dash16PM, 2023-09-19:
@thclark @kersting I think the depth of discussion, above, crosses the bridge from "simple" to "complex."  I think we want to produce a standard that can help with Energy yield analysis, suitability evaluation, and noise impact.  Is somebody wants to document measured power curves or related data using the schema I am ok with it - if there is a demand for this - but as stated earlier ONLY if it is done in a simple fashion - that is the measured average power v wind speed under some specified conditions with a reference to more detailed documentation is the maximum that is captured.  If it gets complicated by taking into consideration the version of the  standard under which the power curve was measured, beyond referencing a report, i would leave this option out of the first edition of the schema.

kersting, 2023-09-19:
@Dash16PM I agree with your comment. I think we should have a mindset of a software for this standard. I think we should deliver a minimum viable product that performs the tasks that most people are performing in energy yield, site suitability, and noise.

I think that adding measured power curves will add some major complications as listed above, and that would not match with the idea of a minimum viable product.

@thclark as we will likely see ourselves working with the complex features, should we step back, and ask ourselves whether it makes sense to work from a software development standpoint and shoot towards an MVP?

thclark, 2023-10-10:
Agree with everyone above! The source of the document can be derived from measurements, and the document metadata section allows descriptions of the source(s) of the data, eg that it's derived from measurements as opposed to simulations or other means. But it'd be a massive scope creep to capture the measurements themselves. Closing this issue.

closed 2023-10-10 by thclark