IEC-61400/power-curve-schema #18: Add identifier for those of us that already have JSON turbine type formats

NicholasMRobinson, 2023-07-06:
## Feature request

We need to be able to branch code on the basis of the first field

### About me

Openwind developer

If you're new to the twined ecosystem, and you don't mind, it'd be great to hear a little bit of the wider context about
you and what you're working on, so we can understand how best to help.

### Use Case

Branch to either our legacy format or this new format

Please [describe your motivation and use case].

### Current state

Right now it looks like you dive straight in. So do we :) We need to continue supporting our users' legacy files while we try to move them onto this new version.

Please describe what you're doing presently to work around this or achieve what you're doing.

NicholasMRobinson, 2023-07-07:
I think that once this identifier is added we would add code to Openwind to track the current octue power curve schema. This would allow us to test on the fly and to add new optional objects that are already compliant with the octue schema.

NicholasMRobinson, 2023-07-19:
Label is really the label for that specific turbine type. What I mean is something to distinguish this JSON file type from any other JSON file type.

Maybe some like 

schema: IEC61400-16

Something that we can easily and reliably if/then on and that will be the same for every single file that uses this schema.

Nick

p.s. sorry for the delay in getting back to you. I am currently travelling with family in Romania.

thclark, 2023-07-27:
@NicholasMRobinson that would normally done in the metadata, by whatever system is managing the files or data blobs, allowing you to identify data **without introspecting the data itself** (otherwise you'd essentially have to read your entire data lake in order to find out which pieces of information you're looking for).

For example, you might have a database with a table of file records for metadata whose columns look like this:
```
id        manufacturer_label        model_name        rated_power    file_path
1         siemens                   SWT-Whatever      6000000        gs://my-bucket/power-curves/siemens/swt-whatever
```

And if you wished to keep track of schema versions explicitly you could add another column (my preference is not to do that but instead to keep all things :
```
id     ....    schema_uri
1      ....    https://jsonschema.registry.octue.com/windpioneers/power-curve/0.0.2.json
```

As a related sidenote, we should avoid property names similar to ones with special meaning like, in this case [the `$schema` dialect property](https://json-schema.org/understanding-json-schema/reference/schema.html#:~:text=The%20%24schema%20keyword%20is%20used,to%20the%20dialect%20%24schema%20identifies..

NicholasMRobinson, 2023-08-12:
I think I fundamentally don't understand what we're doing. I don't get why we would go for the most complicated solution we can think of when there is a simple one. What am I missing please? What I am asking for is clearly completely different from turbine metadata. I want a way to know whether I am reading an Openwind JSON turbine spec or one that is in the format we're all trying to agree on here. Is that info already in there and obvious? Is that why this is causing confusion?

NicholasMRobinson, 2023-08-12:
Is this what we should look for then? Not sure whether you're saying above that you have this or that you're opposed to it.

id     ....    schema_uri
1      ....    https://jsonschema.registry.octue.com/windpioneers/power-curve/0.0.2.json

thclark, 2023-08-24:
Following up on our discussion on this a week (or two?) ago, we decided that we could account for this use case, plus allow much more generally useful document description if we were to allow DCMI metadata entries, and require that at least one of the entries be related to the format of the file. Our consensus was that the DCMI `format` tag was probably the best one.

I've started work on the `document_metadata` section and the [draft of that section can be seen and tried out here](Draft for the document metadata improvements is here: https://codesandbox.io/s/power-curve-document-metadata-dhkjxg?file=/src/schema.json) (this link should take you to a form allowing addition of any number of metadata elements).

I suggest the next step with this is that we add a constraint requiring that at least one `format` element must be added whose value is `IEC61400-16`

Then @NicholasMRobinson's use case is solved out of the box whilst remaining totally compliant with established practices for adding document metadata.

thclark, 2023-09-13:
Done