IEC-61400/power-curve-schema #24: TI distributions for Sales/Warranted Power Curves

thclark, 2023-08-01:
## Feature request

### Use Case

In our meeting of 01 August 2023, in which we were looking at the `design_basis` section, @Dash16PM noted that sometimes you might add a Turbulence Intensity distribution assumed for the development of the Sales/Warrantied Power Curve. 

This isn't the same as the `turbulence_category` (custom or standard) included in the design basis, but does have relevance for advanced site suitability analysis.

We noted that this may be better captured elsewhere in the schema – perhaps as a parameter on which power curves might vary but we're not sure yet.

### Way forward

@Dash16PM we might put a pin in this for now and see if it comes up in review. But if you have any examples or more information to add (such as an explanation how it's different from the turbulence category in the design basis), that might allow us to progress this either by adding a section or deciding to close this.

Dash16PM, 2023-08-01:
The concern I had is relevant to a Developer trying to understand the degree of confidence they can have that a power curve provided by an OEM will accurately represent a turbine in the field. It touches on the realm of "commercial" adjustments to power curves. A simulated power curve provided by an OEM is dependent on a number of inputs, with turbulence intensity (TI) being a critical one.  There are other parameters of interest such as vertical shear, upflow, etc.  But these are not known to be varied, although they could be during the development of a "sales/warrantied" power curve.   There are a number of ways an OEM can vary TI in the definition of their power curves.  They can assume constant TI across all wind speeds. They can assume a low TI below, say 8 m/s, and constant TI above 8 m/s, they can use a standard NTM or an adjusted NTM.  Knowing the assumptions made by an OEM in the development of a "sales/warrantied" power curve can help in the assessment of risk of underperformance (Historically, some OEMs did not take credit for the improved power performance due to high TI in low (below ~8m/s) wind speeds. Sometimes, certain OEMs felt that the reduction in power performance due to TI in the knee of the power curve was over predicted by simulated TI etc.)  In sum, it would be good to know the assumptions that OEMs made in the derivation of their "sales/warrantied" power curves but in the end, current Energy Analysis processes are understood to be conducted using the "sales/warrantied" power curves and power curve adjustments are added to the losses subtracted from the Gross energy that is predicted.  These power curve losses are typically assumed or derived from test results by 3rd parties. A placeholder for assumptions used to develop "sales/warrantied" power curves would be nice to have but not essential to the success of the intent of the -16 standard.  I welcome thoughts from others as to whether or not such a placeholder should be included. (Perhaps we should included it and see if any OEMs feel comfortable populating it.)

thclark, 2024-01-16:
This is now covered by our ability to add custom turbulence distributions.

closed 2024-01-16 by thclark