IEC-61400/power-curve-schema #26: Temperature Derating

Dash16PM, 2023-08-14:
Please see attached a crude draft of what should be added to the Schema.  Redlined edits/suggestions welcome
[Draft.Wind.Turbine.Derating_08142023_Rev0.docx](https://github.com/octue/power-curve-schema/files/12337460/Draft.Wind.Turbine.Derating_08142023_Rev0.docx)

NicholasMRobinson, 2023-08-15:
We currently have three different types of table for temperature derating:

[image]

"Derate":{"Enable":true,"Mode":0,"TempC":[30,35,40],"AirDensity":[1.225,1.1,1],"DeratedMaxPowerKW":[[3000,3000,2000],[3000,2700,1200],[3000,2400,600]]}

[image]

"Derate":{"Enable":true,"Mode":1,"TempC":[30,35,40],"Elevation":[1000,1800,2500],"DeratedMaxPowerKW":[[3000,3000,2000],[3000,2700,1200],[3000,2400,600]]}

[image]

"Derate":{"Enable":true,"Mode":2,"DeratedMaxPowerKW":[3000,2700,2400,2000,600],"Elevation":[1000,1800,2400],"TempC":[[30,37,40,42,45],[30,35,37,40,42],[30,32,35,37,40]]}

NicholasMRobinson, 2023-08-15:
OK, took some hand editing but here they are in the same order in Octueish

"Derate": {
	"Enable": true,
	"Mode": 0,
	"parameters": [
	{
		"label": "temperature C",
		"dimension": 0,
		"points": [30, 35, 40]
	},
	{
		"label": "AirDensity",
		"dimension": 1,
		"points": [1.225, 1.1, 1]
	},
	"DeratedMaxPowerKW": [
		[
			3000000, 3000000, 2000000
		],
		[
			3000000, 2700000, 1200000
		],
		[
			3000000, 2400000, 600000
		]
	]
}

"Derate": {
	"Enable": true,
	"Mode": 1,
	"parameters": [
	{
		"label": "temperature C",
		"dimension": 0,
		"points": [30, 35, 40]
	},
	{
		"label": "Elevation m",
		"dimension": 1,
		"points": [1000, 1800, 2500]
	},
	"DeratedMaxPowerKW": [
		[
			3000000, 3000000, 2000000
		],
		[
			3000000, 2700000, 1200000
		],
		[
			3000000, 2400000, 600000
		]
	]
}

"Derate": {
	"Enable": true,
	"Mode": 2,
	"parameters": [
	{
		"label": "DeratedMaxPower" 
		"dimension": 0,
		"points": [3000000, 2700000, 2400000, 2000000, 600000]
	},
	{
		"label": "Elevation"
		"dimension": 1,
		"points": [1000, 1800, 2400]
	},
	"temperature C": [
		[
			30, 37, 40, 42, 45
		],
		[
			30, 35, 37, 40, 42
		],
		[
			30, 32, 35, 37, 40
		]
	]
}

thclark, 2023-08-29:
To consider:

- Wind Speed derating (with higher wind speed we get more cooling).
- flag to determine whether power derating is a fraction or a max power.

Dash16PM, 2023-09-05:
Please do not forget to define transition between temp set points (linear/other) and re-start lags etc.  See attachment to first comment in this string of comments.

Dash16PM, 2023-09-19:
@thclark I believe we decided during the last call that we do not need to consider wind speed as part of thermal derating "not-required" (at least for the first edition of the standard)

I think it is "essential" to have what Nick laid out but also important "essential" to capture the additional details I added in my draft.

thclark, 2024-02-13:
Update: Kell came back and said don't worry about wind-speed-dependent derating after searching for an example (apparently it was a one-off)

thclark, 2024-06-05:
Captured TODOS from schema drafting session:
```

        // TODO TOM add an explanation about units and why we've chosen constant units into the working draft
        // Describes the thermal operating envelope of the turbine including shutdowns, restarts and progressive derating including the effects of altitude or density
        "thermal_shutdown": {
          "type": "object",
          "required": [],
          "examples": [
            {
              // Cold temperature limits
              "cold": {
                // TODO Make derating optional so we can have simple shutdown

                // Define progressive (linear or nonlinear) deratings as the
                // shutdown temperature is approached. This should contain a
                // conservative estimate of the maximum power produced by the
                // turbine under these conditions (noting that system complexity
                // is high and many different components within the device have different
                // thermal limits and hystereses). Multiple deratings can be provided for
                // multiple air densities or altitudes.
                "derating": [
                  {
                    "$comment": "Example of using altitude instead of air_density",
                    // Altitude in m above WGS84 spheroid [m]. Altitude serves as a proxy
                    // for air density which should be determined from altitude according
                    // to ISO Standard 2533:1975.
                    "altitude": 1743,
                    // Array of temperature values [degrees Centigrade] for which maximum
                    // power limits are given, approaching the shutdown temperature. The
                    // final element of the array should correspond to the shutdown
                    // temperature. At intermediate values of temperature, a linear
                    // interpolation of power is assumed.
                    "temperature": [10, -15],
                    // Array of maximum power [W] values produced by the turbine at
                    // corresponding values in the temperature array, at a given altitude
                    // or air_density.
                    "power_limit": [2e6, 1.1e6]
                  },
                  {
                    "$comment": "Example of a linear profile",
                    // Air density [kg/m^3] at which this derating profile applies.
                    "air_density": 1.0,
                    "temperature": [10, -15],
                    "power_limit": [2e6, 1.1e6]
                  },
                  {
                    "air_density": 1.225,
                    "temperature": [10, -15],
                    "power_limit": [2e6, 1.1e6]
                  }
                ],
                // Put in description that this should meet the endpoint of the derate ranges
                "shutdown_temperature": -15,
                "restart_temperature": -12,
                "restart_duration": 2
              },
              "hot": {
                "derating": [
                  {
                    "air_density": 1.0,
                    "$comment": "Example of a nonlinear profile",
                    "temperature": [40, 41, 42, 43, 44, 45],
                    "power_limit": [2e6, 1.95e6, 1.85e6, 1.7e6, 1.4e6, 0.8e6]
                  },
                  {
                    "air_density": 1.225,
                    "temperature": [40, 41, 42, 43, 44, 45],
                    "power_limit": [2e6, 1.95e6, 1.85e6, 1.7e6, 1.4e6, 0.8e6]
                  }
                ],
                "shutdown_temperature": 45,
                "restart_temperature": 40,
                "restart_duration": 0
              }
            }
          ]
        },
        ```

closed 2024-07-26 by thclark