IEC-61400/power-curve-schema #31: Clarify meaning of "standard edition" in design basis

thclark, 2023-09-07:
## Support request

### Use Case

When creating the design basis notes for #9 we noted down a set of values for a certification (scheme, number, level, revision).

We also said that essential for suitability evaluation is `Edition of Standard (2,3,4 other)` but didn't put down a proper description of what this variable is.

@Dash16PM please can you give (or ping someone to give) a description of this? Is it actually one of the values in the certification code, or does it refer to another kind of standard?

[image]

Dash16PM, 2023-09-07:
The numbers, 2, 3, 4 refer to the editions of IEC 64100-1 the wind turbine design standard. "Other" is if someone states for example Ed.3 Amendment 1 OR if the certification scheme is something other than IEC

I expect to see mostly Ed 2, 3, or 4 and anything different we can chase up directly with the OEM -- so maybe the options are limited to 2, 3, 4, or "contact OEM"

thclark, 2023-09-07:
Thanks @Dash16PM.

Is the design standard a different concept than the certification scheme? I'm trying to work out where it belongs.

Currently we have data that looks like:
```js
design_basis = {
  "certification": {
    "scheme": "IEC",
    "number": "12345",
    "level": "D",
    "revision": "2"
  },
  "standard_edition": "2",
   //... other design basis stuff including class 
}
```

But perhaps the design standard should be more explicitly able to be something other than IEC? Like this?
```
{
  "certification": {
    "scheme": "IEC",
    "number": "12345",
    "level": "D",
    "revision": "2"
  },
  "standard": {
    "body": "IEC",
    "number": "61400-1",
    "edition": "2"
  },
   //... other design basis stuff including class 
}
```

Or perhaps there's another variant I haven't thought of (eg should it be coupled to the class information?

Dash16PM, 2023-09-07:
I try to be precise in the way I refer to things but it is not easy sometimes.  The IEC 61400 series are standards related to wind energy. The series has individual standards focused on different things - these have evolved over time. IEC 61400-1 refers to design of wind turbines. Within this (61400-1) is the design basis. In addition to standards there is a system for the assessment of conformity of a design/turbine model to particular standards - I refer to the guidelines related to assessing conformity OR elaborating on how a standard should be interpreted as "schemes."   Muddying the waters is the fact that IEC developed a standard for the evaluation of conformity IEC 61400-22 (this was formally removed as a Standard BUT it is still being used).  The responsibility for conformity assessment was put under IECRE, which modeled something they refer to as an Operating Document "OD" for the evaluation of conformity.  OD 501 was derived from IEC 61400-22.   

OEMs design hardware
Financing/insuring/countries and maybe a few other entities demand hardware designed by OEMs conforms to a particular standard.  Most standards used in the wind business are produced by IEC. 
Verification that the hardware was designed according to a particular standard produced by a particular recognized entity (IEC, UL, GL/DNV, DiBT, ANSI, ASME, other - In the case of wind, in most cases it is IEC but GL creeps in for older models as can DNV, and in Germany DiBT ) is done by a Cerification Body (such as DNV, UL, CGCC in China etc.) Verification is called "conformity assessment" - Conformity of what is stated in the documentation supporting a certificate (Type Cert [TC] or Design Evaluation Conformity Statement [DECS] or Statement of Compliance [SofC] etc.) This conformity assessment follows some minimum requirements laid out in what used to be referred to as a scheme e.g. OD501 is the most appropriate but we still see IEC 61400-22, and I believe there are others such as legacy GL schemes etc. 

SO 

I had suggested that we capture the number "Certificate Number" or "Registration Number" that appears on the front of an issued Type Cert/DECS/other - for traceability.

The problem is these numbers are not consistent in format.

I would call the number as a whole (all colors above) the "certificate_reference" 

The numerical piece, "pink," I would call "number_part" or "numerical_identifier" or "certificate_numerical_identifier"

The green part I would call "system" or "certification_system" (this is hard because the examples provided refer to the certification standard system and the conformity assessment system - in retrospect, neither are schemes although the IECRE system implies the OD501 scheme was followed for an onshore wind turbine type certificate) 

Blue remains level - but I would call it "certification_level" to make it more precise.

Yellow I would call "certificate_revision"

Finally, I would use something like the following:

{
  "certificate_reference": {
    "certification_system" OR "system": "IEC",
    "certificate_numerical_identifier" OR "numerical_identifier": "12345",
    "certification_level" OR "level": "D",
    "certificate_revision" OR "revision": "2"
  },
   "design_standard" OR "standard": {
    "certification_system" OR "system": "IEC",
    "standard_number" OR "number": "61400-1",
    "standard_edition" OR "edition": "2"
  },

thclark, 2024-01-16:
Closing as completed

closed 2024-01-16 by thclark