IEC-61400/power-curve-schema #37: Decide where to enter allowable grid frequencies.

thclark, 2023-09-13:
## Support request

### Use Case

In #32 Dan Graham and I decided that it would be sensible to have a list of compatible grid frequencies.

The schema would include a property like this:
```js
//...
"grid_frequencies": {
  "type": "array",
  "items": {
    "type": "number"
  },
  "title": "Grid frequencies [Hz]",
  "description": "The design grid frequencies in Hz"
},
```
With matching data looking like this:
```js
{
  "grid_frequencies": [50, 60]
}
```

We already have that earmarked to put under design basis (where it might end up becoming more complicated including tolerances).

From my experiences with power curve documents, this information usually goes fairly front and centre so I think that in the `turbine` section (which is for high-level information) we should have:
```
"compatible_grid_frequencies": {
  "type": "array",
  "items": {
    "type": "number"
  },
  "title": "Compatible grid frequencies [Hz]",
  "description": "A list of compatible/available nominal grid frequencies in Hz"
},
```

Whilst in the design basis we should have *(or possibly more complex to allow a tolerance about each value)*:
```
"grid_frequencies": {
  "type": "array",
  "items": {
    "type": "number"
  },
  "title": "Grid frequencies [Hz]",
  "description": "Grid frequencies in Hz for this design basis. This list must be a subset of (or the same as) the compatible_grid_frequencies in the turbine section".
},
```

@Dash16PM who should I ping to discuss this?

Dash16PM, 2023-09-13:
Firstly, I am unclear what value there is in providing any kind of tolerance around 50Hz or 60hz.  I know in the real world there are tolerances around grid frequency but this kind of detail is way out of scope.  Grid frequency is part of the turbine definition.  Placing it under design basis may not be the best place -- on reflection.  It is probably on the same level as turbine rotor diameter - or just below.  You can have Class II B turbines that have been evaluated under both 50 and 60 Hz -- so I think you are correct. Grid frequency should be under turbine and removed from Design basis.  The schema should not be complicated by tolerances since OEMs typically do not specify these in association with power curves AND I do not believe end users worry about such details for Energy Yield analysis, suitability or noise impact.

thclark, 2023-10-24:
Decision made:

1. This should be renamed `nominal_grid_frequencies` to clarify that it's not a tolerance range. 
2. This should go into the `turbine` section (lots of discussion about whether it should belong in design basis, and whether we should add tolerance but there doesn't seem to be a use case arising for that even for high-level site suitability)

closed 2024-02-13 by thclark