IEC-61400/power-curve-schema #39: Add multiple cut out conditions and hysteresis time

thclark, 2023-09-18:
## Feature request

### Use Case

I've recently seen a cutin/out/recutin scenario in which the following conditions are given for cut out:

> The wind turbine generator system will shut down when the average wind speed exceeds:
> - 25 m/s average in a 600 second time interval, or
> - 30 m/s average in a 30 second time interval, or
> - 34 m/s average in a 3-second time interval
>
> After the turbine has been shut down,  it will restart when the 4-minute  rolling average wind speed drops below 22 m/s.

We should update the schema (probably at the same time as fixing #21) to allow for multiple cases of cutin/cutout/recutin each with a different averaging period and hysteresis period.

thclark, 2023-09-25:
@kersting there's an existing lump in the schema but I'm not sure it comprehensively covers scenarios.
https://github.com/octue/power-curve-schema/blob/db3e708f87797561186ebe7203e4d91ac09a7c25/power-curve-schema/schema.json#L422C14-L422C15

Dash16PM, 2023-09-25:
@thclark  @kersting I skimmed the "lump" and saw one thing I'd like to change, see following code 

        "model_name": {
          "type": "string",
          "title": "Model name",
          "description": "Name of the turbine model (according to the manufacturer)",

Rather than "according to the manufacturer" I recommend we use "as stated on the certificate, TC, DECS, other"

Where TC = Type Certificate
DECS = Design Evaluation Conformity Statement
other could include Statement of Compliance etc.

I am pressing this point because OEMs, in my experience, sometimes use a shorthand or are inconsistent. Within my organization they like to label power to the first decimal point etc.  I press everyone in my loop to use the name on the certification document.  I hope we can do the same here.

Dash16PM, 2023-09-25:
@thclark @kersting  Perhaps we should gather all the descriptions and ask some of the less active team members to review them against things like WindIO and other wind terminology standards/references.

thclark, 2023-09-25:
Data might look somehting like this:
```js
{ "cuts": [
             {
              "cut_in": 3,
              "cut_out": 25,
              "period": 600,
              "recut_in": 22,
              "recut_period": 240,
            },
            {
             "cut_in": 3,
             "cut_out": 32,
             "period": 30,
             "recut_in": 22,
             "recut_period": 240,
           }
          ]
}
```

thclark, 2023-10-10:
NB closing this should also close #21

thclark, 2024-01-26:
Reassigning myself to progress this one

closed 2024-02-13 by thclark