IEC-61400/power-curve-schema #51: definition of classes within Design Basis

Dash16PM, 2024-01-11:
I recommend the following code be modified to reflect this proposed labeling system. standard, optional and special are either common terms in the industry or directly quoted from the -1 design standard.

Proposed labeling system:

Property | Description |

standard_class_label | Select from predefined IEC design classes I, II, III, and T |  
optional_class_label | Cold Climate or "CC" design class |  
special_class_label | Class S, where design values are chosen by the designer |  

Area of Code to be modified:

          "design_class": {
            "title": "Design class",
            "description": "A fully or partially predefined environmental class, (eg IEC or other scheme)",
            "type": "object",
            "oneOf": [
              { 
                "type": "object",
                "required": ["label"],
                "properties": {
                  "label": {
                    "$comment": "TODO change this from `label` to `class_label`",
                    "description": "Select from predefined IEC design classes I, II, and III.",
                    "enum": ["I", "II", "III"]
                  }     
                },
                "$comment": "S (T, CC to be confirmed) is a custom class, use other objects."
              },
              { 
                "type": "object",
                "required": ["label"],
                "properties": {
                  "label": {
                    "$comment": "TODO change this from `label` to `class_label`",
                    "title": "Label",
                    "description": "Specify IEC classes S, T, CC or an entirely custom design class",
                    "enum": ["S", "T", "CC", "Custom"]

thclark, 2024-01-16:
@Dash16PM would the `optional_class_label` `CC` be given *in addition to*, or *instead of* the standard/custom label?

thclark, 2024-01-16:
I want to clear up my question above before acting but I don't think that changing the name of the label key is a good idea here for usability (we don't want user to have to iterate through all three keys just to show the class label) but we could add a 'const' property called `class_kind` which has a set value, so you'd have for example:
```js
{ 
  "design_class": {
    "class_label": "II",
    "class_kind": "standard"
    //...
}
```
or
```js
{ 
  "design_class": {
    "class_label": "S",
    "class_kind": "special"
    //...
}
```

Honestly though overall I think this is redundant data to have, since the value of `class_kind` is directly determined by the value of the `class_label` while being less specific.

thclark, 2024-01-16:
@Dash16PM to illustrate what I mean about usability... If you wanted to print out the value of the label, from (say) python, using a consistent property name the code would be:
```py
print(your_document['design_bases'][0]['design_class']['label'])
```
Whereas with different property names it'd be:
```py
design_class = your_document['design_bases'][0]['design_class']
if 'standard_class_label' in design_class.keys():
   print(design_class['standard_class_label'])
elif 'optional_class_label' in design_class.keys():
   print(design_class['optional_class_label'])
else:
   print(design_class['special_class_label'])
```

closed 2024-03-22 by thclark