IEC-61400/power-curve-schema #52: Labeling of Turbulence Distributions

Dash16PM, 2024-01-11:
The current schema has the following labeling
             "type": "object",
                "description": "Specify Normal and Extreme Turbulence Models (NTM, ETM) as a function of wind speed.",
                "required": ["category", "wind_speed", "normal_turbulence_intensity", "extreme_turbulence_intensity"],

I recommend that normal_turbulence_intensity be renamed normal_turbulence_model, and extreme_turbulence_intensity be renamed extreme_turbulence_model

this will align with the IEC standard and be clear to OEMs as referring to key design distributions.

thclark, 2024-01-16:
I thought that `normal_turbulence_intensity` and `wind_speed` and `category` together comprised a Normal Turbulence Model? Surely the value that's given is a turbulence intensity value and should be called as such? (whichever way we decide, the description must be updated for consistency, since it's currently not consistent)

thclark, 2024-01-16:
A way of resolving might be to call it `normal_turbulence_model_intensity` but it's a bit wordy

closed 2024-02-13 by thclark