IEC-61400/power-curve-schema #74: Working Draft missing descriptions of acoustic-emission properties and potential mismatch in property names in schema

Dash16PM, 2024-08-30:
## Bug report

Following input from Alex James I drilled down into the latest working draft and found some omissions/possible mismatches.  These are described below: 

In the report Clause 4.6.3.1.2, Table 16 we tabulate “Properties of each Operating mode” and list “acoustic_emissions” there.  BUT in the schema under “operating_modes”  in the list of “required” items “acoustic_emisions” does not appear.  As I write I recall we made this optional. 

In Table 16 we reference Table 19 of 4.6.3.1.5 where we have Properties of acoustic emissions – but I cannot find description of  

“margin”
“weighting”
“frequency” or 
“sound_pressure_level” properties.

It looks like the schema I am looking at on Github has some disconnects with the working draft on rows 1376 potentially through 1560.

### What is the expected behavior?

We need to have the labeling harmonized between the schema and the working draft, plus ensure the working draft has all the properties/sub properties used in the schema.

thclark, 2024-09-19:
We're getting slightly tripped over here by something I've done to reduce duplication in the schema.

As acoustic emissions can be specified as 1/3, full, or total, we'd have to repeat the definitions of `margin`, `weighting`, `wind_speed` and `frequency` over and again, duplicating those definitions.

So, what we do in the schema is we `$define` a term for each of those, which is inlined into the `oneOf` where needed.

So our `oneOf` looks like this:
```js

                "oneOf": [
                  {
                    "title": "Emissions at one-third-octave intervals",
                    "properties": {
                      "margin": {
                        "$ref": "#/$defs/acoustic_emissions_margin"
                      },
                      // ... and the other properties for one-third intervals
                  },
                  {
                    "title": "Emissions at full-octave intervals",
                    "properties": {
                      "margin": {
                        "$ref": "#/$defs/acoustic_emissions_margin"
                      },
                      // ... and the other properties for full intervals
                  },
                  {
                    "title": "Total emissions",
                    "properties": {
                      "margin": {
                        "$ref": "#/$defs/acoustic_emissions_margin"
                      },
                      // ... and the other propoerties for total intervals
                  }
                ]
```

Where you see things like:
```js
                        "$ref": "#/$defs/acoustic_emissions_margin"
```

that's an instruction to in-line the common definition for `margin`. It means "take the definition at `$defs/acoustic_emissions_margin` (which is [here in the scheme](https://github.com/octue/power-curve-schema/blob/2b95e3b853d5f994ad06686e530a156caf298471/power-curve-schema/schema.json#L342) and insert it. This definition looks like this:
```js

    "acoustic_emissions_margin": {
      "title": "Margin",
      "description": "A recommended margin of error [dB] which may be added to the given sound pressure levels for practical use.",
      "type": "number",
      "examples": [0, 2],
      "minimum": 0
    },
```

So the schema is functionally equivalent to the following, except if we need to update the definition we can do so in one place rather than several, and we can keep the schema compact too:

```js

                "oneOf": [
                  {
                    "title": "Emissions at one-third-octave intervals",
                    "properties": {
                      "margin": {
                         "title": "Margin",
                         "description": "A recommended margin of error [dB] which may be added to the given sound pressure levels for practical use.",
                         "type": "number",
                         "examples": [0, 2],
                         "minimum": 0
                      },
                      // ... and the other properties for one-third intervals
                  },
                  {
                    "title": "Emissions at full-octave intervals",
                    "properties": {
                      "margin": {
                         "title": "Margin",
                         "description": "A recommended margin of error [dB] which may be added to the given sound pressure levels for practical use.",
                         "type": "number",
                         "examples": [0, 2],
                         "minimum": 0
                      },
                      // ... and the other properties for full intervals
                  },
                  {
                    "title": "Total emissions",
                    "properties": {
                      "margin": {
                         "title": "Margin",
                         "description": "A recommended margin of error [dB] which may be added to the given sound pressure levels for practical use.",
                        "type": "number",
                        "examples": [0, 2],
                        "minimum": 0
                      },
                      // ... and the other propoerties for total intervals
                  }
                ]
```

The path of the reference is purely referential and not part of the schema. What's happened at some point though is that these references to `acoustic_emissions_margin` have got placed incorrectly into the document. So I've tidied up the property names in the standard draft. No changes to teh schema itself are required.

closed 2024-09-19 by thclark

Dash16PM, 2024-09-20:
Tom,

Thanks for the explanation. I have accepted all your changes.

I did find you left the definition of SPL locked - can you unlock it please?

Excited to have finished this phase.  It will be interesting to see the feedback we get.

Probably speak again in a few months

Thanks for all the hard work

Emil

This e-mail may contain privileged or confidential information. If you are not the intended recipient: (1) you may not disclose, use, distribute, copy or rely upon this message or attachment(s); and (2) please notify the sender by reply e-mail, and then delete this message and its attachment(s). UL Solutions Inc. and its affiliates disclaim all liability for any errors, omissions, corruption or virus in this message or any attachments.