IEC-61400/power-curve-schema #76: Fields for referencing a turbine model and operating mode from a different schema

jonssonchristian, 2024-09-10:
Hi @thclark,

I am working on the development of the IEC 61400-15-2 [EYA DEF](https://github.com/IEC-61400-15/eya_def) schema that covers energy yield assessment (EYA) results and other reporting data. We intend to reference the IEC 61400-16 power curve schema for capturing the details of all power curves used in the EYA. One power curve document according to the power curve schema should be included for each turbine model under consideration.

In the part of the schema that describes EYA scenarios, we need to provide a turbine model and baseline (gross energy) operating mode for each turbine at the wind farm(s) under assessment. This will need to match the names/IDs in the power curve document, so that you can look up the corresponding power curve data. I presume the field (property) `model_name` under `turbine` and the field `label` under `power_curves` and `operating_modes` would be the appropriate ones for this purpose. Before going ahead, I thought I would ask for your view on this, in case you have any suggestions we have not thought of. It is of course important the names and labels are unique, and that should be easy to guarantee in the context of an EYA document. In a larger context, like a database of power curves, I suppose there could end up being more than one power curve document for a model name.

jonssonchristian, 2025-02-08:
We have completed an initial implementation to use the Power Curve Schema as a sub-schema in the EYA DEF to cover turbine model performance specifications as reported in an energy yield assessment. That is based on the assumptions I described above. We would very much welcome feedback on our implementation.

https://github.com/IEC-61400-15/eya_def/pull/66

Dash16PM, 2025-02-18:
You state "I suppose there could end up being more than one power curve document for a model name."  I understand this to be correct. FYI there at least one major OEM that has multiple power curves available for the same model of turbine. In part this has been evolutionary. I have also seen tailoring for site environmental conditions and also for loads management, but the latter is likely to fall under an operating mode.

thclark, 2025-04-29:
@jonssonchristian I think you've got the right approach here, basically, so I'm going to close this, but thought I'd mention that I came across an odd edge case this week, which you maybe hadn't thought about - in which a manufacturer was releasing different documents for the same turbine, but with a different subset of all the available modes.

Irritating but definitely work around-able in this situation for -15 by ensuring that you don't have multiple documents for the same turbine.

As far as people's turbine databases go, I'd expect them to keep track of their own document versions (or use dedicated software to do it for them).

jonssonchristian, 2025-06-17:
Hi @thclark, thank you for your feedback on this and apologies for taking ages to respond. I think it seems like a good idea to include a reference to the power curve document identifier when specifying the power curve that applies to a specific turbine in an energy yield assessment (EYA).

It would also be workable to reference only the turbine model name and the baseline operating mode label, as we have it now, but requiring that an EYA DEF document cannot include multiple Power Curve Schema documents with the same model name. However, that could open up for errors if versions are updated in one place but not another. It seems more robust to explicitly reference a document identifier, which would make it easy to detect any version mismatch. As suggested by @Dash16PM, power curves for a turbine model do often evolve over time.

I just created an [issue](https://github.com/IEC-61400-15/eya_def/issues/86) and [pull request](https://github.com/IEC-61400-15/eya_def/pull/87) on the EYA DEF repo to include a reference to the document identifier value.

closed 2025-09-30 by thclark