IEC-61400/power-curve-schema #77: Would it be relevant to include environmental parameter validity ranges?

jonssonchristian, 2025-02-08:
Hi @thclark,

The schema currently covers reference design values for environmental parameters under the design bases. When we worked on integrating the Power Curve Schema as a sub-schema of the EYA DEF, a member of our group raised the question of whether we can capture validity ranges for environmental parameters, such as turbulence and wind shear (power law exponent).

Whereas the design basis provides representative conditions for which the wind turbine was designed, the validity limits would be the filtering conditions to apply in a power curve measurement test. Some organisations will use validity range data to inform turbine performance adjustment factors in an energy yield assessment.

Many OEMs will list power curve validity ranges in power curve documents. For wind shear it is typically a minimum and maximum exponent valid. For turbulence intensity it may be a minimum and maximum valid, a minimum and maximum curve by wind speed bin, or a minimum and maximum curve expressed as multipliers using the standard IEC turbulence intensity curve.

Have I understood it correctly that validity ranges are outside of the current scope of the schema? If yes, could it be relevant and practical to include it, or have you already taken a conscious decision within the IEC 61400-16 group to leave it outside of the scope? I thought I ask the question just in case this has not already come up and you think it could be a useful addition.

(This question arose after the CD process for the IEC 61400-16 document closed, so I was not able to include it with my comments there.)

thclark, 2025-02-18:
We're discussing the ability to include these maxima and minima in the parameter definitions.

Ideal solution is to include the ability to EITHER include the full range of parameter definitions (expecting the matrix to be full) OR where a single dimension is given, a valid range is supplied.

Key points:
- SGRE noted that the valid range of TI varies with Wind Speed, so the valid ranges must be a function of wind speed

thclark, 2025-04-29:
WIP on this is in #86