IEC-61400/power-curve-schema #78: Review feedback

jonssonchristian, 2025-02-08:
Hi @thclark,

As part of reviewing the CD of the IEC 61400-16 document, I went through the JSON Schema in detail. I tried to put all my feedback in OSD, but had noted a few comments very specific to the JSON Schema implementation, which did not fit there. In case you find it useful, I have listed those below.

- Would it be useful for the `"description"` at the top level to be expanded and aligned with the IEC 61400-16 document?

- Would it be useful to include descriptions under the definitions for `"2darray"`, `"3darray"` and `"4darray"`?

- In an array item under `"design_bases"`, the properties `"standard_climate"`, `"cold_climate"` and `"hot_climate"` all appear to have the same structure. Could a definition be introduced and referenced for each instead of repeating the same schema? Also, the sub-properties `"operating_temperature_range"` and `"survival_temperature_range"` have the same structure, and could perhaps also have a common definition.

- Should the `"operating_temperature_range"` and `"survival_temperature_range"` arrays have `"minItems"` and `"maxItems"` included and both set to 2, and `"uniqueItems"` included set to `true`?

- For the special or custom design class, would it be better to define the type of `"class_label"` as `"const": "S"` rather than `"enum": ["S"]` (i.e. an enum with a single option is a constant)?

- Would it make sense for all strings to have a minimum length of 1, to avoid users encoding the lack of data as an empty string ("")? Some string properties have this already, but not all.

- Could the string regular expression pattern `"^[a-z0-9_]{1,40}$"` be simplified to just `"^[a-z0-9_]+$"` since the length is cover by the max and min length parameters?

- Under each array item under `"operating_modes"`, would it be helpful to include examples under `"parameters"`?

- Why is `"additionalItems"` set to `true` on many of the arrays (for example on the `"parameters"` array and on the `"values"` sub-property array)? In my understanding the `"additionalItems"` keyword is not used in the latest version of the JSON Schema specification.

- I noted the following typos:
  - overridden miswritten as "overriden";
  - exceeded miswritten as  "exceded"; and
  - conformity miswritten as "confirmity".

- I noted there are still some TODO comments and questions to the schema team. Are those items still open?

- For the 2D turbulence model example (see below), would it be clearer to have different values across the wind speed bins rather than repeating the same ones?

```json
"normal_turbulence_intensity": [
  [0.116, 0.1889, 0.2613, 0.3337, 0.46],
  [0.116, 0.1889, 0.2613, 0.3337, 0.46],
  [0.116, 0.1889, 0.2613, 0.3337, 0.46]
]
```

I would be happy to put up a pull request to address some of this, if you agree and think that would be useful.

Thanks for all the great work on this schema and I look forward to it becoming an international standard soon.

Dash16PM, 2025-02-18:
This looks like very valuable feedback - much appreciated.

jonssonchristian, 2025-02-20:
You are very welcome. I hope it will be of use.

When working through an example, which I had originally copied from this repo, I also realised that the `"rated_power"` property was incorrectly included directly under an object in `""operating_modes""`. See for example [here](https://github.com/octue/power-curve-schema/blob/2b95e3b853d5f994ad06686e530a156caf298471/power-curve-schema/examples/generic-117-3.json#L198). I presume that was there in an earlier version of the schema, and then changed to being defined under the turbine object and as overrides in the operating mode objects. It is one downside with allowing additional properties that it is easy to miss if things are unintentionally left in the wrong place.

thclark, 2025-12-19:
> Would it make sense for all strings to have a minimum length of 1, to avoid users encoding the lack of data as an empty string ("")? Some string properties have this already, but not all.

I think this messes with some form libraries which will initialise with an error. In most JS libs, blank is taken as nullish anyway so the actual intent gets captured. Plus, we're very late in the stage now so whilst I think this comment is valid, I probably don't want to touch things in favour of getting the schema out of the door!