IEC-61400/power-curve-schema #79: Usage feedback

olivia-bentley, 2025-03-04:
Hi,

Recently I've been looking at the schema to so I can start integrating it within some new processes we are working on. Firstly, the schema is very well thought and out and easy to follow. It has been straight forward to pick up and begin using. While I've started using I've noticed a couple of things.

There seems to be usage of both "air-density" and "air_density" in the schema at the moment. From the comments I think "air_density" was chosen.

I was wondering if it would be useful to include mode within the temperature derating in addition to air density and altitude options.

When there are different power curves at different hub heights in the schema would you implement these as different modes?

thclark, 2025-04-01:
@olivia-bentley I'm sorry, I didn't get a notification for this, would you be open to a quick call to discuss? If so it'd be great if you could [book a lost](https://calendly.com/thclark/60min?month=2025-04)

thclark, 2025-04-15:
Conclusion of our conversation:

Mode-based derating could be done by adding optional mode_label to the derating specifications, something like this:

```

              {
                "title": "Altitude based derating",
                "required": ["altitude", "power_limit", "temperature"],
                "properties": {
                  "mode_label": {
                    "description": "The mode label for which this derating applies, if missing then assumed to apply to all modes.",
                    "string": "number"
                  },
                  "altitude": {
                    "description": "The altitude at which this derating curve applies [m above the WGS84 spheroid]",
                    "type": "number"
                  },
                  "temperature": {
                    "description": "A list of atmospheric temperature values [degrees C]",
                    "type": "array",
                    "items": {
                      "type": "number"
                    }
                  },
                  "power_limit": {
                    "description": "A list of power limits each corresponding to an atmospheric temperature value in the previous list [W]",
                    "type": "array",
                    "items": {
                      "type": "number"
                    }
                  }
                }
              },
              {
                "title": "Air density based derating",
                "required": ["air_density", "power_limit", "temperature"],
                "properties": {
                  "mode_label": {
                    "description": "The mode label for which this derating applies, if missing then assumed to apply to all modes.",
                    "string": "number"
                  },
                  "air_density": {
                    "description": "The air density at which this derating curve applies [kg/m^3]",
                    "type": "number"
                  },
```

thclark, 2025-04-15:
> There seems to be usage of both "air-density" and "air_density" in the schema at the moment. From the comments I think "air_density" was chosen.

Where `air-density` is used, it's in the context of a label (eg a string property value), where used in snake case `air_density` is part of a variable/property *name*.

> When there are different power curves at different hub heights in the schema would you implement these as different modes?

Yes, you'd use the `overrides` section to delineate a restricted set of `available_hub_heights` for a particular mode. Then you can use multiple modes each with different hub heights.

closed 2025-04-29 by thclark